VladimirYugay/Gaussian-SLAM #4: Question about Table 1  and Table 2 in paper.

lif314, 2024-03-04:
It seems there might be a discrepancy or an error in your paper regarding the results reported in Table 1 for the ScanNet dataset and Table 2 for the TUM-RGBD dataset. The paper mentions that the results for Nice-SLAM[1] and Vox-Fusion[2] are from Nicer-SLAM [3], but I couldn't find relevant experimental results in the Nicer-SLAM paper. Additionally, the Nicer-SLAM paper only includes results for the Replica and 7-Scenes datasets.

[1]  Zihan Zhu, Songyou Peng, Viktor Larsson, Weiwei Xu, Hujun Bao, Zhaopeng Cui, Martin R Oswald, and Marc Pollefeys. Nice-slam: Neural implicit scalable encoding for slam. In IEEE/CVF Conference on Computer Vision and Pattern Recognition, pages 12786–12796, 2022. 1, 2, 6, 7, 8, 9, 10,11.
[2] Xingrui Yang, Hai Li, Hongjia Zhai, Yuhang Ming, Yuqian Liu, and Guofeng Zhang. Vox-fusion: Dense tracking and mapping with voxel-based neural implicit representation. In IEEE International Symposium on Mixed and Augmented Reality (ISMAR), pages 499–507. IEEE, 2022. 2, 6, 7, 8, 10.
[3] Zihan Zhu, Songyou Peng, Viktor Larsson, Zhaopeng Cui, Martin R Oswald, Andreas Geiger, and Marc Pollefeys. Nicerslam: Neural implicit scene encoding for rgb slam. arXiv preprint arXiv:2302.03594, 2023. 2, 6.

VladimirYugay, 2024-03-04:
Hey there, you are completely right. Thanks a lot for pointing this out. 

It is a copy-paste mistake on our writing side. Rendering performance on real-world datasets (TUM_RGB, Scannet) was never reported by previous methods. We pasted this text in all rendering tables hence the error. We will fix it, thanks again!

lif314, 2024-03-04:
> Hey there, you are completely right. Thanks a lot for pointing this out.
>
> It is a copy-paste mistake on our writing side. Rendering performance on real-world datasets (TUM_RGB, Scannet) was never [...]

Thank you for your work! Could you please provide the tracking results of these methods on the ScanNet and TUM-RGBD datasets? Thanks!

VladimirYugay, 2024-03-04:
Hey there, for this work we used DROID-SLAM for our tracking backend. So if you want to track performance on those datasets you can directly check the  [DROID-SLAM](https://arxiv.org/pdf/2108.10869.pdf) performance in tracking.

One more thing, we are about to get permission to release the code (we hope 1 or 2 weeks at most) and plan to do it ASAP. In the released code there will be a new tracking method with new (better) results which will be also reflected on arxiv.

lif314, 2024-03-04:
> Hey there, for this work we used DROID-SLAM for our tracking backend. So if you want to track performance on those datasets you can directly check the [...]

Thanks, looking forward to it! I will close this issue.

closed 2024-03-04 by lif314

zhangshuoneu, 2024-07-11:
> Hey there, for this work we used DROID-SLAM for our tracking backend. So if you want to track performance on those datasets you can directly check the [...]

Hi, Wonderful work! But I haven't noticed that droid-slam is mentioned in paper. If DROID-SLAM is used to tracking, the init camera tracking pose should be provided by droid-slam. I just find the constant speed assumption in paper.  Besides, I also haven't found the droid.pth in git.

VladimirYugay, 2024-07-11:
Hey @zhangshuoneu ,

Thanks for your interest in our work! We made another iteration of the paper to get rid of DROID-SLAM. You can find the updated version of our paper on arXiv.