VladimirYugay/Gaussian-SLAM #6: Depth Supervision

ThePassedWind, 2024-03-25:
Hi, your work is amazing.
Could you provide some clues of the implementation of depth supervision?
I have tried other versions of depth-based `diff-gaussian-rasterization` package, such as:
1. https://github.com/ashawkey/diff-gaussian-rasterization
2. https://github.com/ingra14m/depth-diff-gaussian-rasterization/tree/depth-mean3D

However, the depth supervision for xyz of gaussians is extremely weak!
I'd like to know what's the difference between their package and yours for depth supervision.
Could you help me explain this issue?

VladimirYugay, 2024-03-25:
You can take a look at our implementation if you like (we also described it in the paper). I believe, our implementations are very similar/same.

ThePassedWind, 2024-03-27:
Thanks for your help.

closed 2024-03-27 by ThePassedWind