VladimirYugay/Gaussian-SLAM #14: how to visualize the result？

KwanWaiPang, 2024-03-27:
Dear authors, 
I would like to express my gratitude for your exceptional work.
I am interested in learning more about how I can visualize the reconstruction results. For instance, MonoGS utilizes OpenGL, 3DGS employs SIBR_viewers, and Splatam comes with its own viz_scripts. 
Do you use the "gaussian-splatting-lightning "? Could you please provide some insight into the use of "gaussian-splatting-lightning"? Perhaps you could share some examples in the readme?

Thank you

Atticuszz, 2024-03-27:
search in issues

VladimirYugay, 2024-03-27:
Hey @KwanWaiPang, thanks for your interest! Most of the visualizers use the exported Gaussian model ply file and render it. At the end of the run, during evaluation, we compute both the mesh as @Atticuszz mentioned, and the Gaussian parameters exported in the ply file [here](https://github.com/VladimirYugay/Gaussian-SLAM/blob/a94421893a96a7b600a1e53079af92766472e2a8/src/evaluation/evaluator.py#L220). 

Having this exported Gaussian ply file, you can use pretty much any visualizer you prefer. For example, for gs-splatting-lightning, you can install their repo, and just call `visualize.py <path_to_the_exported_gaussian_model.ply>. After this you can play around with the parameters and the camera trajectory fly-through.

KwanWaiPang, 2024-03-27:
thank you very much！

closed 2024-03-27 by KwanWaiPang