VladimirYugay/Gaussian-SLAM #16: Custom Dataset

hariharan1412, 2024-03-29:
First of all, thank you for this wonderful project.

I have a capture of my house, and is there a way to run my own dataset?

unique1i, 2024-03-29:
Thanks for the interest in our work.

Please prepare a dataset class similar to, e.g. [scannet](https://github.com/VladimirYugay/Gaussian-SLAM/blob/a94421893a96a7b600a1e53079af92766472e2a8/src/entities/datasets.py#L171). The input needs to be RGBD data, with color and depth of same size. You could use our config on scannet or scannet++ to have a try then.

hariharan1412, 2024-03-29:
Could you please help me?
I would be so grateful if you could at least give me a hint. 
thank you

unique1i, 2024-03-29:
We need color_data, depth_data, and poses sorted by index in the new dataset class. To run on the scene, the config needs new camera [intrinsics](https://github.com/VladimirYugay/Gaussian-SLAM/blob/a94421893a96a7b600a1e53079af92766472e2a8/configs/ScanNet/scannet.yaml#L35) and dataset [path](https://github.com/VladimirYugay/Gaussian-SLAM/blob/a94421893a96a7b600a1e53079af92766472e2a8/configs/ScanNet/scene0000_00.yaml#L2), also add new dataset into the [get_dataset](https://github.com/VladimirYugay/Gaussian-SLAM/blob/a94421893a96a7b600a1e53079af92766472e2a8/src/entities/datasets.py#L261) function. You could test data loading with debugger first.

hariharan1412, 2024-03-29:
Thank you so much, I'll close this issue

closed 2024-03-29 by hariharan1412

ddenglina, 2024-07-15:
> We need color_data, depth_data, and poses sorted by index in the new dataset class. To run on the scene, the config needs new camera [...]

I have color_data,depth_data. How can I get poses?