VladimirYugay/Gaussian-SLAM #20: Installation Failed !!

hariharan1412, 2024-04-15:
conda env update --file environment.yml 
Collecting package metadata (repodata.json): | WARNING conda.models.version:get_matcher(546): Using .* with relational operator is superfluous and deprecated and will be removed in a future version of conda. Your spec was 1.8.0.*, but conda is ignoring the .* and treating it as 1.8.0
WARNING conda.models.version:get_matcher(546): Using .* with relational operator is superfluous and deprecated and will be removed in a future version of conda. Your spec was 1.6.0.*, but conda is ignoring the .* and treating it as 1.6.0
WARNING conda.models.version:get_matcher(546): Using .* with relational operator is superfluous and deprecated and will be removed in a future version of conda. Your spec was 1.7.1.*, but conda is ignoring the .* and treating it as 1.7.1
WARNING conda.models.version:get_matcher(546): Using .* with relational operator is superfluous and deprecated and will be removed in a future version of conda. Your spec was 1.9.0.*, but conda is ignoring the .* and treating it as 1.9.0
done
Solving environment: done

==> WARNING: A newer version of conda exists. <==
  current version: 23.3.1
  latest version: 24.3.0

Please update conda by running

    $ conda update -n base -c conda-forge conda

Or to minimize the number of packages updated during conda update use

     conda install conda=24.3.0

Installing pip dependencies: | Ran pip subprocess with arguments:
['/home/ubuntu/conda/envs/gslam/bin/python', '-m', 'pip', 'install', '-U', '-r', '/home/ubuntu/workdir/Gaussian-SLAM/condaenv.rg07_ib3.requirements.txt', '--exists-action=b']
Pip subprocess output:
Collecting git+https://github.com/eriksandstroem/evaluate_3d_reconstruction_lib.git@9b3cc08be5440db9c375cc21e3bd65bb4a337db7 (from -r /home/ubuntu/workdir/Gaussian-SLAM/condaenv.rg07_ib3.requirements.txt (line 10))
  Cloning https://github.com/eriksandstroem/evaluate_3d_reconstruction_lib.git (to revision 9b3cc08be5440db9c375cc21e3bd65bb4a337db7) to /tmp/pip-req-build-k2z9n_op
  Resolved https://github.com/eriksandstroem/evaluate_3d_reconstruction_lib.git to commit 9b3cc08be5440db9c375cc21e3bd65bb4a337db7
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
Collecting git+https://github.com/VladimirYugay/simple-knn.git@c7e51a06a4cd84c25e769fee29ab391fe5d5ff8d (from -r /home/ubuntu/workdir/Gaussian-SLAM/condaenv.rg07_ib3.requirements.txt (line 11))
  Cloning https://github.com/VladimirYugay/simple-knn.git (to revision c7e51a06a4cd84c25e769fee29ab391fe5d5ff8d) to /tmp/pip-req-build-r9sin004
  Resolved https://github.com/VladimirYugay/simple-knn.git to commit c7e51a06a4cd84c25e769fee29ab391fe5d5ff8d
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
Collecting git+https://github.com/VladimirYugay/gaussian_rasterizer.git@9c40173fcc8d9b16778a1a8040295bc2f9ebf129 (from -r /home/ubuntu/workdir/Gaussian-SLAM/condaenv.rg07_ib3.requirements.txt (line 12))
  Cloning https://github.com/VladimirYugay/gaussian_rasterizer.git (to revision 9c40173fcc8d9b16778a1a8040295bc2f9ebf129) to /tmp/pip-req-build-vh3i8h9e
  Resolved https://github.com/VladimirYugay/gaussian_rasterizer.git to commit 9c40173fcc8d9b16778a1a8040295bc2f9ebf129
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
Requirement already satisfied: open3d==0.18.0 in /home/ubuntu/conda/envs/gslam/lib/python3.10/site-packages (from -r /home/ubuntu/workdir/Gaussian-SLAM/condaenv.rg07_ib3.requirements.txt (line 1)) (0.18.0)
Requirement already satisfied: wandb in /home/ubuntu/conda/envs/gslam/lib/python3.10/site-packages (from -r /home/ubuntu/workdir/Gaussian-SLAM/condaenv.rg07_ib3.requirements.txt (line 2)) (0.16.6)
Collecting trimesh (from -r /home/ubuntu/workdir/Gaussian-SLAM/condaenv.rg07_ib3.requirements.txt (line 3))
  Using cached trimesh-4.3.0-py3-none-any.whl.metadata (18 kB)
Collecting pytorch_msssim (from -r /home/ubuntu/workdir/Gaussian-SLAM/condaenv.rg07_ib3.requirements.txt (line 4))
  Using cached pytorch_msssim-1.0.0-py3-none-any.whl.metadata (8.0 kB)
Collecting torchmetrics (from -r /home/ubuntu/workdir/Gaussian-SLAM/condaenv.rg07_ib3.requirements.txt (line 5))
  Using cached torchmetrics-1.3.2-py3-none-any.whl.metadata (19 kB)
Requirement already satisfied: tqdm in /home/ubuntu/conda/envs/gslam/lib/python3.10/site-packages (from -r /home/ubuntu/workdir/Gaussian-SLAM/condaenv.rg07_ib3.requirements.txt (line 6)) (4.66.2)
Requirement already satisfied: imageio in /home/ubuntu/conda/envs/gslam/lib/python3.10/site-packages (from -r /home/ubuntu/workdir/Gaussian-SLAM/condaenv.rg07_ib3.requirements.txt (line 7)) (2.34.0)
Requirement already satisfied: opencv-python in /home/ubuntu/conda/envs/gslam/lib/python3.10/site-packages (from -r /home/ubuntu/workdir/Gaussian-SLAM/condaenv.rg07_ib3.requirements.txt (line 8)) (4.9.0.80)
Requirement already satisfied: plyfile in /home/ubuntu/conda/envs/gslam/lib/python3.10/site-packages (from -r /home/ubuntu/workdir/Gaussian-SLAM/condaenv.rg07_ib3.requirements.txt (line 9)) (1.0.3)
Requirement already satisfied: numpy>=1.18.0 in /home/ubuntu/conda/envs/gslam/lib/python3.10/site-packages (from open3d==0.18.0->-r /home/ubuntu/workdir/Gaussian-SLAM/condaenv.rg07_ib3.requirements.txt (line 1)) (1.26.4)
Requirement already satisfied: dash>=2.6.0 in /home/ubuntu/conda/envs/gslam/lib/python3.10/site-packages (from open3d==0.18.0->-r /home/ubuntu/workdir/Gaussian-SLAM/condaenv.rg07_ib3.requirements.txt (line 1)) (2.16.1)
Requirement already satisfied: werkzeug>=2.2.3 in /home/ubuntu/conda/envs/gslam/lib/python3.10/site-packages (from open3d==0.18.0->-r /home/ubuntu/workdir/Gaussian-SLAM/condaenv.rg07_ib3.requirements.txt (line 1)) (3.0.2)
Requirement already satisfied: nbformat>=5.7.0 in /home/ubuntu/conda/envs/gslam/lib/python3.10/site-packages (from open3d==0.18.0->-r /home/ubuntu/workdir/Gaussian-SLAM/condaenv.rg07_ib3.requirements.txt (line 1)) (5.10.4)
Requirement already satisfied: configargparse in /home/ubuntu/conda/envs/gslam/lib/python3.10/site-packages (from open3d==0.18.0->-r /home/ubuntu/workdir/Gaussian-SLAM/condaenv.rg07_ib3.requirements.txt (line 1)) (1.7)
Requirement already satisfied: ipywidgets>=8.0.4 in /home/ubuntu/conda/envs/gslam/lib/python3.10/site-packages (from open3d==0.18.0->-r /home/ubuntu/workdir/Gaussian-SLAM/condaenv.rg07_ib3.requirements.txt (line 1)) (8.1.2)
Requirement already satisfied: addict in /home/ubuntu/conda/envs/gslam/lib/python3.10/site-packages (from open3d==0.18.0->-r /home/ubuntu/workdir/Gaussian-SLAM/condaenv.rg07_ib3.requirements.txt (line 1)) (2.4.0)
Requirement already satisfied: pillow>=9.3.0 in /home/ubuntu/conda/envs/gslam/lib/python3.10/site-packages (from open3d==0.18.0->-r /home/ubuntu/workdir/Gaussian-SLAM/condaenv.rg07_ib3.requirements.txt (line 1)) (9.4.0)
Requirement already satisfied: matplotlib>=3 in /home/ubuntu/conda/envs/gslam/lib/python3.10/site-packages (from open3d==0.18.0->-r /home/ubuntu/workdir/Gaussian-SLAM/condaenv.rg07_ib3.requirements.txt (line 1)) (3.8.4)
Requirement already satisfied: pandas>=1.0 in /home/ubuntu/conda/envs/gslam/lib/python3.10/site-packages (from open3d==0.18.0->-r /home/ubuntu/workdir/Gaussian-SLAM/condaenv.rg07_ib3.requirements.txt (line 1)) (2.2.2)
Requirement already satisfied: pyyaml>=5.4.1 in /home/ubuntu/conda/envs/gslam/lib/python3.10/site-packages (from open3d==0.18.0->-r /home/ubuntu/workdir/Gaussian-SLAM/condaenv.rg07_ib3.requirements.txt (line 1)) (6.0.1)
Requirement already satisfied: scikit-learn>=0.21 in /home/ubuntu/conda/envs/gslam/lib/python3.10/site-packages (from open3d==0.18.0->-r /home/ubuntu/workdir/Gaussian-SLAM/condaenv.rg07_ib3.requirements.txt (line 1)) (1.4.2)
Requirement already satisfied: pyquaternion in /home/ubuntu/conda/envs/gslam/lib/python3.10/site-packages (from open3d==0.18.0->-r /home/ubuntu/workdir/Gaussian-SLAM/condaenv.rg07_ib3.requirements.txt (line 1)) (0.9.9)
Requirement already satisfied: Click!=8.0.0,>=7.1 in /home/ubuntu/conda/envs/gslam/lib/python3.10/site-packages (from wandb->-r /home/ubuntu/workdir/Gaussian-SLAM/condaenv.rg07_ib3.requirements.txt (line 2)) (8.1.7)
Requirement already satisfied: GitPython!=3.1.29,>=1.0.0 in /home/ubuntu/conda/envs/gslam/lib/python3.10/site-packages (from wandb->-r /home/ubuntu/workdir/Gaussian-SLAM/condaenv.rg07_ib3.requirements.txt (line 2)) (3.1.43)
[... cut ...]
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for gaussian_rasterizer
ERROR: Could not build wheels for simple_knn, gaussian_rasterizer, which is required to install pyproject.toml-based projects

failed

CondaEnvException: Pip failed

nvcc --version

nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2022 NVIDIA Corporation
Built on Wed_Jun__8_16:49:14_PDT_2022
Cuda compilation tools, release 11.7, V11.7.99
Build cuda_11.7.r11.7/compiler.31442593_0

unique1i, 2024-04-15:
Hi @hariharan1412, are you updating the conda environment or doing fresh install? The provided yaml file will avoid cuda mismatch error.

hariharan1412, 2024-04-15:
i tried both none of them is working, I'm keep on getting this error.

unique1i, 2024-04-15:
I can't reproduce this error, the conda env is tested locally and on cluster.

hariharan1412, 2024-04-15:
Could you please share your env list with me so I can create mine from there?

unique1i, 2024-04-15:
This won't solve the problem, as gaussian_rasterizer and simple_knn need to be built. The error message above shows cuda mismatch:
```
The detected CUDA version (11.7) mismatches the version that was used to compile PyTorch (10.2)
```
, but none of these versions are listed in our yaml file, as we use pytorch 2.1 with cuda 12.1. Please double check.

hariharan1412, 2024-04-15:
I have already successfully setup Gaussian Splatting conda environment, which also requires both Gaussian_rasterizer and simple_knn in the same system.

unique1i, 2024-04-15:
I see, please use a separate conda environment for our repo to avoid any cuda mismatch with the current environment.

VladimirYugay, 2024-04-16:
@hariharan1412, as an alternative, try installing the Gaussian Splatting environment (which you already can do), and then install the requirements mentioned in our environment file without specifying their versions. In the very end, install our rasterizer.

maliksyria, 2024-04-26:
Updating conda solved the issue.

closed 2024-04-26 by unique1i