VladimirYugay/Gaussian-SLAM #21: About depthL1 metric

anghaocheng, 2024-05-13:
Hello, thanks for sharing this excellent article; I could get the normal psnr and F1 metric, but I didn't know how to read the depth L1 result; In the output file, **reconstruction_metrics.json** and **rendering_metrics.json** record depth l1 and depth l1_train_view indicators, respectively. What do they mean

anghaocheng, 2024-05-13:
By the way, when I ran the 3D rebuild evaluation,open3d gave a lot of warnings and ended up with a much larger depth l1 in **reconstruction_metrics.json** than in **rendering_metrics.json**, is there any problem?
[image]

unique1i, 2024-05-15:
Hi @anghaocheng, thanks for your interest in our work.

`depth l1_train_view` is the averaged rendered depth loss compared to the input RGBD frames that we used to train the mapper; it does not measure the reconstructed mesh. `depth l1` measures the rendered depth using reconstructed mesh at newly sampled views compared to using ground truth mesh. It is expected that `depth l1_train_view` will be smaller.

I comitted a fix to address this Open3D visualizer issue. Please help check again and note that our reported `depth_l1` values are obtained with headless version of Open3D.

closed 2024-05-31 by unique1i