VladimirYugay/Gaussian-SLAM #23: About the initialization of the pose for tracking optimization on Replica dataset

pengchongH, 2024-05-28:
Hi! Thanks for your nice work!

I wonder that the reason why you choose multi-scale RGBD odometry to initialize the pose on Replica rather than using constant speed assumption which is used in other datasets. Could you please share some brief idea behind this choice?

unique1i, 2024-05-29:
Hi @pengchongH. Constant speed assumption works nicely also on Replica dataset. The odometry initialization here gives slightly better performance.

pengchongH, 2024-05-30:
Thanks for your reply. I have tried to use constant speed on the Replica dataset. However, it looks like the constant speed cannot work well in room0 and room1, but it works well in office0.
[image]

unique1i, 2024-05-31:
I confirm the const speed initialization alone does lead to large pose error on `room0` and `room1`. The is due to that a few times along the sequence there are much higher re-rendering error after initialization. 

A mechanism is there to switch to odometer initialization in this case, please also enable `help_camera_initialization` in `configs/Replica/replica.yaml` when using const speed initialization. This helps get small pose error.

pengchongH, 2024-06-02:
Thanks for your reply. I have tried this and it works well. I think this issue only occurs in room scenes, including room0-room2, in Replica, since you use const speed assumption for other datasets, and the const speed assumption also works well for office scenes in Replica. Based on this, I would like to know whether you have some brief idea why this kind of issue occurs.

VladimirYugay, 2024-06-13:
Pose optimization is quite sensitive for initialization. We think that in the rooms since they fall to the local minima and can't recover, while odom initialization helps avoid them.

VladimirYugay, 2024-06-26:
@pengchongH After investigation, we figured out that there was a big in the release branch. This [function](https://github.com/VladimirYugay/Gaussian-SLAM/blob/ce8c887252dea89f74f96f5904266636b1253beb/src/utils/tracker_utils.py#L63) works, but not entirely correct - we need to initialize rotation quaternions differently. We fixed it and now testing it on all the datasets. We will push the fix to the main branch.

VladimirYugay, 2024-07-03:
Pushed to the main. Feel free to reopen if you have any further questions!

closed 2024-07-03 by VladimirYugay

pengchongH, 2024-07-04:
Thanks for your reply. I will try your updated codes. Really appreciate!