VladimirYugay/Gaussian-SLAM #26: Wrong quaternion manipulation?

smileyenot983, 2024-06-13:
Thanks for this great work on gaussian slam. 
I found the code, which performs rotation extrapolation in def `interpolate_poses`:
Here rotation extrapolation is done the same way as for translation, however, I suppose that quaternions could not be simply added and subtracted:
`init_rot = quat_poses[1] + (quat_poses[1] - quat_poses[0])`

I suppose, that quaternion mathematics should be used here() and it should be multiplication instead of sum, as well as multiplication by conjugate instead of subtraction, something like this:
`init_rot = quat_poses[1] * (quat_poses[1] * quat_poses[0].inv())`  

What do you think about this?

VladimirYugay, 2024-06-17:
Hey there. Thanks for spotting this and sorry for the late reply. We will investigate this deeper with additional experiments and be back to you. Currently, we are a bit busy, but in a week it should be better.

VladimirYugay, 2024-06-27:
Hey there, you are right. We missed a bug when we released the code. We are now running additional experiments, but the function should look like:

```
def extrapolate_poses(poses: np.ndarray) -> np.ndarray:
    """ Generates an interpolated pose based on the first two poses in the given array.
    Args:
        poses: An array of poses, where each pose is represented by a 4x4 transformation matrix.
    Returns:
        A 4x4 numpy ndarray representing the interpolated transformation matrix.
    """
    return poses[1, :] @ np.linalg.inv(poses[0, :]) @ poses[1, :]
```

VladimirYugay, 2024-07-03:
Pushed to the main, feel free to re-open if there are any other questions, and thanks for spotting this!

closed 2024-07-03 by VladimirYugay