VladimirYugay/Gaussian-SLAM #28: Run other code?

YIXIN-M, 2024-06-28:
How did I obtain the values of evaluation metrics such as PNSR, SSIM, and LPIPS for NICE-SLAM, Vox Fusion, ESLAM, Point SLAM, and SplaTAM * in the paper? I downloaded the corresponding source code from GitHub and ran it on my computer, but I couldn't obtain the specific values of these evaluation metrics.

VladimirYugay, 2024-06-28:
Hey there. Thanks for your interest in our work. Some of the metrics we took from this [survey](https://arxiv.org/abs/2402.13255). Concurrent methods we ran ourselves. To obtain results we ran each method three times with the seeds 0, 1, 2 and averaged them out.

YIXIN-M, 2024-06-28:
These ESLAM, Splatam, nice slam can run but cannot obtain specific values for evaluation metrics such as PNSR, SSIM, and LPIPS in papers. Point slam can obtain these values in the Replica dataset, but cannot in other datasets. Do you want to add code for evaluation metrics such as PNSR, SSIM, and LPIPS to these source codes?

VladimirYugay, 2024-06-28:
We didn't plan to add rendering evaluation code for other datasets.

VladimirYugay, 2024-07-08:
Closing it for now. Feel free to re-open if you have any other questions!

closed 2024-07-08 by VladimirYugay