VladimirYugay/Gaussian-SLAM #29: About the initialization of 3D gaussian

pengchongH, 2024-07-06:
Thanks for your nice work. I noticed that you use the following code to initialize the position of 3D Gaussian. I am curious about why you choose 1.005 * depth rather than 1 * depth. Because the GT depth is an important clue to initialize the position of 3D Gaussian. Could you please give some brief ideas related to this choice?

pts = create_point_cloud(gt_color, 1.005 * gt_depth, intrinsics, estimate_c2w)

VladimirYugay, 2024-07-08:
Hey there, thanks for your interest in our work. We found empirically, that when we initialize the Gaussians a little bit further from the surface it gives us slightly better results. We assume that this happens because there's more room for the means of the Gaussians to move during the optimization. However, this gives only marginal performance gain and can be considered as a hack.

VladimirYugay, 2024-07-16:
Feel free to re-open the issue if you have any other questions!

closed 2024-07-16 by VladimirYugay