VladimirYugay/Gaussian-SLAM #30: [question] which files are necessary for running this repo's model if I use my own data?

zydmtaichi, 2024-07-08:
Suppose I can get the rgb and depth imgs in separate folder by using a rgbd camera, but what i should do to get the rest txt files like GT and accelerometer? Or maybe other txt files can be omitted? please give me more detailed info to explain how to run gaus-slam on my own data

VladimirYugay, 2024-07-08:
Hey there, thanks for your interest in our work.

We use Txt files from the datasets to measure our tracking performance. To run the method on your data, you don't need them. 

What you need are 2 folders with RGB and depth images, and camera intrinsic parameters for them. Also, you have to define the scaling parameter for your depth measurements `depth_scale`.

zydmtaichi, 2024-07-11:
> Hey there, thanks for your interest in our work.
>
> We use Txt files from the datasets to measure our tracking performance. To run the method on your data, you don't need them.
>
> What you need are 2 [...]

Thanks for your reply, i will follow the advise above and try it later.

closed 2024-07-11 by zydmtaichi

ddenglina, 2024-07-17:
> Thanks for your reply, i will follow the advise above and try it later.

Hey, Have you try? Omitted pose is available?