VladimirYugay/Gaussian-SLAM #31: Question about the visualization reconstruction scene

Yoona12, 2024-07-14:
Thanks for your great work!
When seeing the video, it has a  picture of visualization reconstruction scene as following.
[image]

I visualize the reconstruction scene by SuperSplat, and the result is shown following.
[image]
There is a roof in the visualization of the scene，so i am confused about how to remove the roof and get the the visualization reconstruction scene as yours?
Looing forward to your reply!

unique1i, 2024-07-16:
Hi, @Yoona12,

this is correct, when the fly-through video is made, the roof part is cropped for visualization purpose. You can use select and crop in supersplat to do the cropping.

VladimirYugay, 2024-07-24:
Feel free to re-open if you have any other questions!

closed 2024-07-24 by VladimirYugay