VladimirYugay/Gaussian-SLAM #32: > We need color_data, depth_data, and poses sorted by index in the new dataset class. To run on the scene, the config needs new camera [intrinsics](https://github.com/VladimirYugay/Gaussian-SLAM/blob/a94421893a96a7b600a1e53079af92766472e2a8/configs/ScanNet/scannet.yaml#L35) and dataset [path](https://github.com/VladimirYugay/Gaussian-SLAM/blob/a94421893a96a7b600a1e53079af92766472e2a8/configs/ScanNet/scene0000_00.yaml#L2), also add new dataset into the [get_dataset](https://github.com/VladimirYugay/Gaussian-SLAM/blob/a94421893a96a7b600a1e53079af92766472e2a8/src/entities/datasets.py#L261) function. You could test data loading with debugger first.

ddenglina, 2024-07-15:
> We need color_data, depth_data, and poses sorted by index in the new dataset class. To run on the scene, the config needs new camera [intrinsics](https://github.com/VladimirYugay/Gaussian-SLAM/blob/a94421893a96a7b600a1e53079af92766472e2a8/configs/ScanNet/scannet.yaml#L35) and dataset [path](https://github.com/VladimirYugay/Gaussian-SLAM/blob/a94421893a96a7b600a1e53079af92766472e2a8/configs/ScanNet/scene0000_00.yaml#L2), also add new dataset into the [get_dataset](https://github.com/VladimirYugay/Gaussian-SLAM/blob/a94421893a96a7b600a1e53079af92766472e2a8/src/entities/datasets.py#L261) function. You could test data loading with debugger first.

I have color_data,depth_data. How can I get poses?

_Originally posted by @ddenglina in https://github.com/VladimirYugay/Gaussian-SLAM/issues/16#issuecomment-2227588785_

VladimirYugay, 2024-07-15:
Hey there,

Poses are originally needed to measure how accurate the pose estimation is. If you are running it on your own data, you can just put identity matrices instead of the pose matrices and run your algorithm.

If you want to measure the tracking performance, you can try something like COLMAP to obtain the poses for your sequence and then use them in our pipeline.

ddenglina, 2024-07-16:
> Hey there,
> 
> Poses are originally needed to measure how accurate the pose estimation is. If you are running it on your own data, you can just put `np.eye(4)` matrices instead of the pose matrices and run your algorithm.
> 
> If you want to measure the tracking performance, you can try something like COLMAP to obtain the poses for your sequence and then use them in our pipeline.

Thanks for your reply.

（1）Do you mean that both gt_pose and estimated_pose are np.eye(4)?

（2）Colmap calculates pose based on RGB data and lacks scale information. Is this available?

VladimirYugay, 2024-07-16:
1. Yes. `estimated_pose` will be filled with values during the run of the algorithm
2. You are right, but I think the poses estimated with colmap might still be a good proxy measure pose estimation quality.

VladimirYugay, 2024-07-24:
Feel free to re-open if you have any further questions!

closed 2024-07-24 by VladimirYugay