VladimirYugay/Gaussian-SLAM #37: refine phase gpu?

chenqi13814529300, 2024-07-31:
Hello, I found that the video memory occupied in the renfine stage is huge, running room0 under the default configuration will burst the video memory, more than 20G, how to optimize?

VladimirYugay, 2024-07-31:
Hey there. 

Yes, there are several ways how to significantly reduce GPU consumption.

1. If you run the method from scratch, you can set this [parmeter](https://github.com/VladimirYugay/Gaussian-SLAM/blob/eaec10d73ce7511563882b8856896e06d1f804e3/configs/Replica/replica.yaml#L12) to 60K instead of 600k. And also reduce `new_submap_gradient_points_num` from 50K to 5K. This will not have a big impact on the performance on Replica, but will significantly speed up the method and lower GPU consumption.

2. Another option is to sparse the number of Gaussians used in the merged map. For example, [here](https://github.com/VladimirYugay/Gaussian-SLAM/blob/eaec10d73ce7511563882b8856896e06d1f804e3/src/evaluation/evaluate_merged_map.py#L73), you can create a subsample mask, and prune the GaussianModel with it such that there are less of them left. If you do this for every submap, GPU consumption will be reduced significantly. The rendering quality will not suffer much.

Hope this helps!

chenqi13814529300, 2024-08-01:
Thank you for your specific answer. I compared SplaTAM MonoGS、Gaussian-SLAM、RTG-SLAM， Considering various aspects such as effectiveness, generalization ability, and learning cycle, we ultimately plan to use your work as a framework for learning. Can we use lidar, IMU, and camera (not depth cameras) to improve your framework for better rendering effects and speed. I am currently engaged in some research in this area and hope to receive some suggestions. Thank you very much.

chenqi13814529300, 2024-08-01:
Thank you for your answer. I will study hard

VladimirYugay, 2024-08-01:
Feel free to re-open if you have any further questions!

closed 2024-08-01 by VladimirYugay

chenqi13814529300, 2024-08-04:
The next question may be a bit silly:
1. Does the accuracy of depth maps have a significant impact on training
2. If we obtain point clouds for each frame, can we only convert this point cloud into a depth map