VladimirYugay/Gaussian-SLAM #38: Would it work with RGB input and depth from mono depth estimator (proxy depth)?

LaFeuilleMorte, 2024-08-22:
Hi, thanks for open source this brilliant work! I have some rgb images but I don't have lidar to get real depth. Is it possible for me to utilize some mono depth estimator (eg. Depth Anything v2 ) to estimate depth for the rgb images ? And therefore do the SLAM with your methods? Additionally, I don't have camera poses either. But maybe I could use colmap or some SFM meseasures to get the poses. Were these possible for doing slam with your methods. Thanks in advance!

VladimirYugay, 2024-08-22:
Hey there, thanks for your interest in our work.

1. Yes, you can try using the depth maps from an external model. However, we didn't test it in that scenario. Just be careful with how you load the depth maps in the data loader (usually our method expects depth in meters which is not always true in off-the-shelf depth estimation models)
2. We use camera poses only for evaluation purposes. You don't really need them to run the algorithm. The easy solution might be just to set them as `np.eye(4)` in the dataloader if you don't have them.

VladimirYugay, 2024-09-11:
Feel free to re-open if you have any further questions!