VladimirYugay/Gaussian-SLAM #39: Can you please clarify the transformations?

JzHuai0108, 2024-09-03:
Dear author,
Thanks you for open sourcing the code. I read it and got baffled by the transformations.
Can you please clarify the below questions?

Note here I use the notation A_T_B which satisfies p^A = A_T_B * p^B.

From the function in_frustum, I think that the viewmatrix's transpose, V', will transform the mean position of a Gaussian to the current camera frame.
Let's assume the Gaussians in the cuda renderer are in a frame, F.
so I think the V matrix is (currentcam_T_F)'. 

These mean 3D positions in F are transformed from the Gaussian model in frame G
by a transformation matrix F_T_G which is to be optimized by the tracker as shown in compute_losses() function.

In rendering settings of the track() function, the viewmatrix is set to be V = (last_w2c)'.
so last_w2c = currentcam_T_F = reference_w2c per the [line](https://github.com/VladimirYugay/Gaussian-SLAM/blob/main/src/entities/tracker.py#L137).

Now let's look [line](https://github.com/VladimirYugay/Gaussian-SLAM/blob/main/src/entities/tracker.py#L213),
it states that
final_c2w = W_T_currentcam = (reference_w2c * best_w2c)^{-1} = (currentcam_T_F * F_T_G) ^{-1} = G_T_currentcam.
So everything looks consistent so far, and W = G, and G seems like a fixed world frame.

However, there are a few confusing point.
1. what is the real meaning of F? a reference camera frame or the current camera frame?
3. From my early experience, the optimized variable should be currentcam_T_G, thus F should be the current camera frame. But then why do we use reference_w2c?

VladimirYugay, 2024-12-02:
Hey there. Thanks for your interest in our work! 

We are not sure we fully understood your question. 

Instead of estimating the current camera pose T_j directly, we optimize the relative camera pose T_{j-1,j} of frame j  with respect to j - 1. To achieve the equivalent of rendering at the current pose T_j, we transform the submap with the relative transformation T_{j-1,j}^{-1} and render from the last camera pose T_{j-1} to get the rendered color \hat{I}_j and depth \hat{D}_j.

As empirical proof that our implementation works, you can try disabling the optimization, and use just the linear pose interpolation and you will notice a significant decrease in tracking accuracy.

Hope this helps and feel free to ping if you have any further questions.

closed 2025-01-07 by VladimirYugay

JzHuai0108, 2025-01-09:
Sorry for the late reply. Thank you for the clarification. I will take time to go through the transformation flow.