VladimirYugay/Gaussian-SLAM #40: Error when evaluating Depth-L1

EmeryLee97, 2024-09-13:
Hi, I got the following error when running reconstruction evaluation:
```
Error = [ GLFW Error: X11: The DISPLAY environment variable is missing 
```
I tried to create a virtual screen using xvfb, but still failed, how did you avoid this error?

VladimirYugay, 2024-09-16:
Hey there. Thanks for your interest in our work.

I think that you're trying to do this on a remote machine without a display. In that case, to finish TSDF fusion, you need a headless open3d version. Alternatively, run the reconstruction on a local machine with the display.

EmeryLee97, 2024-09-21:
> Hey there. Thanks for your interest in our work.
> 
> I think that you're trying to do this on a remote machine without a display. In that case, to finish TSDF fusion, you need a headless open3d version (usually super tedious to install). Alternatively, run the reconstruction on a local machine with the display.

Yes, it is super tedious to install, and it's even more tedious since I'm running on a remote server with no sudo permission. But thanks for your reply!

closed 2024-09-21 by EmeryLee97