VladimirYugay/Gaussian-SLAM #41: *** stack smashing detected ***: terminated

Howie-Ye, 2024-10-04:
Hi, I got the following error when running the code:
I have strictly followed the configuration in the environment.yml, and I used the dataset from HuggingFace as you provided.

(gslam) root@autodl-container-880e41b20a-aeacaa52:~/autodl-tmp/Gaussian-SLAM# python run_slam.py configs/Replica/room0.yaml --input_path data/Replica-SLAM/room0/ --output_path output/Replica/room0
Loaded 2000 frames
Tracking config
{'alpha_thre': 0.98,
 'cam_rot_lr': 0.0002,
 'cam_trans_lr': 0.002,
 'filter_alpha': False,
 'filter_outlier_depth': True,
 'gt_camera': False,
 'help_camera_initialization': False,
 'init_err_ratio': 5,
 'iterations': 60,
 'mask_invalid_depth': False,
 'odometer_method': 'point_to_plane',
 'odometry_type': 'odometer',
 'soft_alpha': True,
 'w_color_loss': 0.95}
Mapping config
{'alpha_thre': 0.6,
 'current_view_opt_iterations': 0.4,
 'iterations': 100,
 'map_every': 5,
 'new_frame_sample_size': -1,
 'new_points_radius': 1e-07,
 'new_submap_every': 50,
 'new_submap_gradient_points_num': 50000,
 'new_submap_iterations': 1000,
 'new_submap_points_num': 600000,
 'pruning_thre': 0.1,
 'submap_using_motion_heuristic': True}

Mapping frame 0
Number of added points:  617145
*** stack smashing detected ***: terminated
Aborted (core dumped)

Howie-Ye, 2024-10-18:
> Hey there! Thanks for your interest in our work. I've never encountered this problem before, but can you check whether the Gaussians fit into your GPU at the very first step?

Thank you for response. It appears that certain functions in Open3D are throwing errors when I working on a remote server without a display attached.

VladimirYugay, 2024-10-18:
Thanks for spotting it! Feel free to re-open if you have any other questions.

closed 2024-10-18 by VladimirYugay