VladimirYugay/Gaussian-SLAM #45: the results are not as expected.

goldtimes, 2025-01-20:
in the bash
python run_slam.py configs/TUM_RGBD/rgbd_dataset_freiburg1_desk.yaml --input_path /media/kilox/PS2000/photo_slam_data/data/TUM/rgbd_dataset_freiburg1_desk --output_path output/TUM/desk

[image]

VladimirYugay, 2025-01-20:
Thanks for your interest in our work! 

Can you please report the ATE and PSNR obtained on this sequence?

goldtimes, 2025-01-21:
here is ATE.json and PSNR
{"compared_pose_pairs": 592, "rmse": 0.062046918602192254, "mean": 0.05791550874710083, "median": 0.05207182094454765, "std": 0.022262418642640114, "min": 0.0, "max": 0.0988922268152237}

[image]

goldtimes, 2025-01-21:
do you mean rendering_metrics.json file? 
here it's contents 
{"psnr": 24.996692822847752, "lpips": 0.14584543960804194, "ssim": 0.9370739935450472, "depth_l1_train_view": 0.18504035514142356, "num_renders": 519}

VladimirYugay, 2025-01-21:
Yes. I think the results reported in Table 2 accurately match your results. So I'd say that's the expected result you got!

VladimirYugay, 2025-01-26:
Feel free to re-open if you have any further questions!

closed 2025-01-26 by VladimirYugay