VladimirYugay/Gaussian-SLAM #46: Questions about learning rate

lee12332, 2025-02-18:
Thanks for your excellent work.
I found that the learning rate used in the “mapper” and “tracker” is always the initial  learning rate obtained from config, rather than the exponential decay learning rate strategy in 3DGS. Why do we do this and what are the benefits?

VladimirYugay, 2025-02-18:
Thanks for your interest in our work!

We wanted to simplify the pipeline as much as possible, and we didn't notice that the decay brought any improvement. Intuitively, we do not need decay because we optimize every sub-map for a relatively small number of iterations compared to full 3DGS. It might be useful for the final refinement but we did not ablate over this.

lee12332, 2025-02-18:
Thank you very much for your answer

closed 2025-02-18 by lee12332