VladimirYugay/Gaussian-SLAM #47: Error while creating conda env from yml file

MAli7319, 2025-03-09:
Hi. I wanted to try out your intriguing work on my computer. (NVIDIA RTX 3050 and Ubuntu 22.04 LTS) However I encountered a very long error message. I could get some points of it and tried different setups but nonetheless they did not work. Can you check it out please?

 `/home/masenturk/anaconda3/lib/python3.12/argparse.py:2006: FutureWarning: `remote_definition` is deprecated and will be removed in 25.9. Use `conda env create --file=URL` instead.
  action(self, namespace, argument_values, option_string)
Channels:
 - pytorch
 - nvidia
 - nvidia/label/cuda-12.1.0
 - conda-forge
 - defaults
Platform: linux-64
Collecting package metadata (repodata.json): done
Solving environment: done

Downloading and Extracting Packages:
                                                                                
Preparing transaction: done
Verifying transaction: done
Executing transaction: - 
| 
done
Installing pip dependencies: - Ran pip subprocess with arguments:
['/home/masenturk/anaconda3/envs/gslam/bin/python', '-m', 'pip', 'install', '-U', '-r', '/home/masenturk/Gaussian-SLAM/condaenv.uh9xh_le.requirements.txt', '--exists-action=b']
Pip subprocess output:
Collecting git+https://github.com/eriksandstroem/evaluate_3d_reconstruction_lib.git@9b3cc08be5440db9c375cc21e3bd65bb4a337db7 (from -r /home/masenturk/Gaussian-SLAM/condaenv.uh9xh_le.requirements.txt (line 10))
  Cloning https://github.com/eriksandstroem/evaluate_3d_reconstruction_lib.git (to revision 9b3cc08be5440db9c375cc21e3bd65bb4a337db7) to /tmp/pip-req-build-_ugsieu4
  Resolved https://github.com/eriksandstroem/evaluate_3d_reconstruction_lib.git to commit 9b3cc08be5440db9c375cc21e3bd65bb4a337db7
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
Collecting git+https://github.com/VladimirYugay/simple-knn.git@c7e51a06a4cd84c25e769fee29ab391fe5d5ff8d (from -r /home/masenturk/Gaussian-SLAM/condaenv.uh9xh_le.requirements.txt (line 11))
  Cloning https://github.com/VladimirYugay/simple-knn.git (to revision c7e51a06a4cd84c25e769fee29ab391fe5d5ff8d) to /tmp/pip-req-build-3wt_1re_
  Resolved https://github.com/VladimirYugay/simple-knn.git to commit c7e51a06a4cd84c25e769fee29ab391fe5d5ff8d
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
Collecting git+https://github.com/VladimirYugay/gaussian_rasterizer.git@9c40173fcc8d9b16778a1a8040295bc2f9ebf129 (from -r /home/masenturk/Gaussian-SLAM/condaenv.uh9xh_le.requirements.txt (line 12))
  Cloning https://github.com/VladimirYugay/gaussian_rasterizer.git (to revision 9c40173fcc8d9b16778a1a8040295bc2f9ebf129) to /tmp/pip-req-build-_z0avyz2
  Resolved https://github.com/VladimirYugay/gaussian_rasterizer.git to commit 9c40173fcc8d9b16778a1a8040295bc2f9ebf129
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
Collecting open3d==0.18.0 (from -r /home/masenturk/Gaussian-SLAM/condaenv.uh9xh_le.requirements.txt (line 1))
  Using cached open3d-0.18.0-cp310-cp310-manylinux_2_27_x86_64.whl.metadata (4.2 kB)
Collecting wandb (from -r /home/masenturk/Gaussian-SLAM/condaenv.uh9xh_le.requirements.txt (line 2))
  Using cached wandb-0.19.8-py3-none-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (10 kB)
Collecting trimesh (from -r /home/masenturk/Gaussian-SLAM/condaenv.uh9xh_le.requirements.txt (line 3))
  Using cached trimesh-4.6.4-py3-none-any.whl.metadata (18 kB)
Collecting pytorch_msssim (from -r /home/masenturk/Gaussian-SLAM/condaenv.uh9xh_le.requirements.txt (line 4))
  Using cached pytorch_msssim-1.0.0-py3-none-any.whl.metadata (8.0 kB)
Collecting torchmetrics (from -r /home/masenturk/Gaussian-SLAM/condaenv.uh9xh_le.requirements.txt (line 5))
  Using cached torchmetrics-1.6.2-py3-none-any.whl.metadata (20 kB)
Collecting tqdm (from -r /home/masenturk/Gaussian-SLAM/condaenv.uh9xh_le.requirements.txt (line 6))
  Using cached tqdm-4.67.1-py3-none-any.whl.metadata (57 kB)
Collecting imageio (from -r /home/masenturk/Gaussian-SLAM/condaenv.uh9xh_le.requirements.txt (line 7))
  Using cached imageio-2.37.0-py3-none-any.whl.metadata (5.2 kB)
Collecting opencv-python (from -r /home/masenturk/Gaussian-SLAM/condaenv.uh9xh_le.requirements.txt (line 8))
  Using cached opencv_python-4.11.0.86-cp37-abi3-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (20 kB)
Collecting plyfile (from -r /home/masenturk/Gaussian-SLAM/condaenv.uh9xh_le.requirements.txt (line 9))
  Using cached plyfile-1.1-py3-none-any.whl.metadata (2.1 kB)
Requirement already satisfied: numpy>=1.18.0 in /home/masenturk/anaconda3/envs/gslam/lib/python3.10/site-packages (from open3d==0.18.0->-r /home/masenturk/Gaussian-SLAM/condaenv.uh9xh_le.requirements.txt (line 1)) (1.26.4)
Collecting dash>=2.6.0 (from open3d==0.18.0->-r /home/masenturk/Gaussian-SLAM/condaenv.uh9xh_le.requirements.txt (line 1))
  Using cached dash-2.18.2-py3-none-any.whl.metadata (10 kB)
Collecting werkzeug>=2.2.3 (from open3d==0.18.0->-r /home/masenturk/Gaussian-SLAM/condaenv.uh9xh_le.requirements.txt (line 1))
  Using cached werkzeug-3.1.3-py3-none-any.whl.metadata (3.7 kB)
Collecting nbformat>=5.7.0 (from open3d==0.18.0->-r /home/masenturk/Gaussian-SLAM/condaenv.uh9xh_le.requirements.txt (line 1))
  Using cached nbformat-5.10.4-py3-none-any.whl.metadata (3.6 kB)
Collecting configargparse (from open3d==0.18.0->-r /home/masenturk/Gaussian-SLAM/condaenv.uh9xh_le.requirements.txt (line 1))
  Using cached ConfigArgParse-1.7-py3-none-any.whl.metadata (23 kB)
Collecting ipywidgets>=8.0.4 (from open3d==0.18.0->-r /home/masenturk/Gaussian-SLAM/condaenv.uh9xh_le.requirements.txt (line 1))
  Using cached ipywidgets-8.1.5-py3-none-any.whl.metadata (2.3 kB)
Collecting addict (from open3d==0.18.0->-r /home/masenturk/Gaussian-SLAM/condaenv.uh9xh_le.requirements.txt (line 1))
  Using cached addict-2.4.0-py3-none-any.whl.metadata (1.0 kB)
[... 434 lines cut ...]
      
      cuda_rasterizer/rasterizer_impl.cu(175): warning #549-D: variable "img" is used before its value is set
         obtain(chunk, img.n_contrib, N, 128);
                       ^
      
      Remark: The warnings can be suppressed with "-diag-suppress <warning-number>"
      
      cuda_rasterizer/rasterizer_impl.cu(183): warning #549-D: variable "binning" is used before its value is set
         obtain(chunk, binning.point_list, P, 128);
                       ^
      
      16 errors detected in the compilation of "cuda_rasterizer/rasterizer_impl.cu".
      error: command '/home/masenturk/anaconda3/envs/gslam/bin/nvcc' failed with exit code 2
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for gaussian_rasterizer
ERROR: Failed to build installable wheels for some pyproject.toml based projects (simple_knn, gaussian_rasterizer)

failed

CondaEnvException: Pip failed
`

At the simple knn building wheel part, the errors are beginning to print out. As i said, i could not solve it. Maybe the reason is the previous steps i can not be sure.

MAli7319, 2025-03-10:
they are both 11.4.0

VladimirYugay, 2025-03-11:
Can you try doing the following:
```
# CUDA 11.8
conda install pytorch==2.5.1 torchvision==0.20.1 pytorch-cuda=11.8 -c pytorch -c nvidia
# CUDA 12.1
conda install pytorch==2.5.1 torchvision==0.20.1 pytorch-cuda=12.1 -c pytorch -c nvidia
# CUDA 12.4
conda install pytorch==2.5.1 torchvision==0.20.1 pytorch-cuda=12.4 -c pytorch -c nvidia
```

Then, try installing `simple-knn` only (if it installs, then it will be easy to install the rasterizer):
```
pip install git+https://github.com/VladimirYugay/simple-knn.git@c7e51a06a4cd84c25e769fee29ab391fe5d5ff8d
```

If that doesn't work, it almost certainly means an issue with the `gcc` version.

MAli7319, 2025-03-11:
I tried with cuda 12.1 because in environment.yml file, cuda versions are indicated as 12.1. So we expect to install cuda-compiler with version of 12.1, but instead it installs 12.8. I thought this is the problem. Because the errors are stem from some cuda functions, and maybe some parts are changed.

The error message turned out to this:
`
Collecting git+https://github.com/VladimirYugay/simple-knn.git@c7e51a06a4cd84c25e769fee29ab391fe5d5ff8d
  Cloning https://github.com/VladimirYugay/simple-knn.git (to revision c7e51a06a4cd84c25e769fee29ab391fe5d5ff8d) to /tmp/pip-req-build-rn_k_4v1
  Running command git clone --filter=blob:none --quiet https://github.com/VladimirYugay/simple-knn.git /tmp/pip-req-build-rn_k_4v1
  Running command git rev-parse -q --verify 'sha^c7e51a06a4cd84c25e769fee29ab391fe5d5ff8d'
  Running command git fetch -q https://github.com/VladimirYugay/simple-knn.git c7e51a06a4cd84c25e769fee29ab391fe5d5ff8d
  Resolved https://github.com/VladimirYugay/simple-knn.git to commit c7e51a06a4cd84c25e769fee29ab391fe5d5ff8d
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [8 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/pip-req-build-rn_k_4v1/setup.py", line 13, in <module>
          from torch.utils.cpp_extension import CUDAExtension, BuildExtension
        File "/home/masenturk/anaconda3/envs/gslam/lib/python3.10/site-packages/torch/__init__.py", line 367, in <module>
          from torch._C import *  # noqa: F403
      ImportError: libcusparse.so.12: cannot open shared object file: No such file or directory
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
`

I found the 'libcusparse.so.12' path and did `export` `LD_LIBRARY_PATH=` but it also did not work.

MAli7319, 2025-03-11:
2.5.1 True

But i could not get the point. Should not we install pytorch and other utils after we activate the gslam env?

VladimirYugay, 2025-03-11:
Yes, activate before installing!

MAli7319, 2025-03-12:
Ok. After that should I install dependencies in yml file by hand?

MAli7319, 2025-03-12:
I needed to install cuda-compiler 12.4 to prevent the version mismatch. Afterwards, I have tried to install simpleknn and it has given this error message.

> Collecting git+https://github.com/VladimirYugay/simple-knn.git@c7e51a06a4cd84c25e769fee29ab391fe5d5ff8d
  Cloning https://github.com/VladimirYugay/simple-knn.git (to revision c7e51a06a4cd84c25e769fee29ab391fe5d5ff8d) to /tmp/pip-req-build-gqvm6hzv
  Running command git clone --filter=blob:none --quiet https://github.com/VladimirYugay/simple-knn.git /tmp/pip-req-build-gqvm6hzv
  Running command git rev-parse -q --verify 'sha^c7e51a06a4cd84c25e769fee29ab391fe5d5ff8d'
  Running command git fetch -q https://github.com/VladimirYugay/simple-knn.git c7e51a06a4cd84c25e769fee29ab391fe5d5ff8d
  Resolved https://github.com/VladimirYugay/simple-knn.git to commit c7e51a06a4cd84c25e769fee29ab391fe5d5ff8d
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: simple_knn
  Building wheel for simple_knn (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [29 lines of output]
      running bdist_wheel
      /home/masenturk/miniconda3/envs/gslam/lib/python3.11/site-packages/torch/utils/cpp_extension.py:497: UserWarning: Attempted to use ninja as the BuildExtension backend but we could not find ninja.. Falling back to using the slow distutils backend.
        warnings.warn(msg.format('we could not find ninja.'))
      running build
      running build_ext
      /home/masenturk/miniconda3/envs/gslam/lib/python3.11/site-packages/torch/utils/cpp_extension.py:361: UserWarning:
      
                                     !! WARNING !!
      
      !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
      Your compiler (/home/masenturk/miniconda3/envs/gslam/bin/x86_64-conda-linux-gnu-c++) is not compatible with the compiler Pytorch was
      built with for this platform, which is g++ on linux. Please
      use g++ to to compile your extension. Alternatively, you may
      compile PyTorch from source using /home/masenturk/miniconda3/envs/gslam/bin/x86_64-conda-linux-gnu-c++, and then you can also use
      /home/masenturk/miniconda3/envs/gslam/bin/x86_64-conda-linux-gnu-c++ to compile your extension.
      
      See https://github.com/pytorch/pytorch/blob/master/CONTRIBUTING.md for help
      with compiling PyTorch from source.
      !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
      
                                    !! WARNING !!
      
        warnings.warn(WRONG_COMPILER_WARNING.format(
      /home/masenturk/miniconda3/envs/gslam/lib/python3.11/site-packages/torch/utils/cpp_extension.py:426: UserWarning: There are no /home/masenturk/miniconda3/envs/gslam/bin/x86_64-conda-linux-gnu-c++ version bounds defined for CUDA version 12.4
        warnings.warn(f'There are no {compiler_name} version bounds defined for CUDA version {cuda_str_version}')
      building 'simple_knn._C' extension
      creating build/temp.linux-x86_64-cpython-311
      /home/masenturk/miniconda3/envs/gslam/bin/x86_64-conda-linux-gnu-c++ -fvisibility-inlines-hidden -std=c++17 -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /home/masenturk/miniconda3/envs/gslam/include -I/home/masenturk/miniconda3/envs/gslam/targets/x86_64-linux/include -L/home/masenturk/miniconda3/envs/gslam/targets/x86_64-linux/lib -L/home/masenturk/miniconda3/envs/gslam/targets/x86_64-linux/lib/stubs -DNDEBUG -D_FORTIFY_SOURCE=2 -O2 -isystem /home/masenturk/miniconda3/envs/gslam/include -I/home/masenturk/miniconda3/envs/gslam/targets/x86_64-linux/include -L/home/masenturk/miniconda3/envs/gslam/targets/x86_64-linux/lib -L/home/masenturk/miniconda3/envs/gslam/targets/x86_64-linux/lib/stubs -fPIC -I/home/masenturk/miniconda3/envs/gslam/lib/python3.11/site-packages/torch/include -I/home/masenturk/miniconda3/envs/gslam/lib/python3.11/site-packages/torch/include/torch/csrc/api/include -I/home/masenturk/miniconda3/envs/gslam/lib/python3.11/site-packages/torch/include/TH -I/home/masenturk/miniconda3/envs/gslam/lib/python3.11/site-packages/torch/include/THC -I/home/masenturk/miniconda3/envs/gslam/include -I/home/masenturk/miniconda3/envs/gslam/include/python3.11 -c ext.cpp -o build/temp.linux-x86_64-cpython-311/ext.o -DTORCH_API_INCLUDE_EXTENSION_H -DPYBIND11_COMPILER_TYPE=\"_gcc\" -DPYBIND11_STDLIB=\"_libstdcpp\" -DPYBIND11_BUILD_ABI=\"_cxxabi1011\" -DTORCH_EXTENSION_NAME=_C -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++17
      error: command '/home/masenturk/miniconda3/envs/gslam/bin/x86_64-conda-linux-gnu-c++' failed: No such file or directory
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for simple_knn
  Running setup.py clean for simple_knn
Failed to build simple_knn
ERROR: Failed to build installable wheels for some pyproject.toml based projects (simple_knn)

It seems, as you said, the issue is related with g++ version. Which version is required?

VladimirYugay, 2025-03-12:
We use `gcc` and `g++` of version 10 locally. For setting things up on Ubuntu, I recommend following this [guide](https://linuxconfig.org/how-to-switch-between-multiple-gcc-and-g-compiler-versions-on-ubuntu-20-04-lts-focal-fossa).

VladimirYugay, 2025-03-12:
Typically, if you switch your gcc version, you need to create an environment from scratch and install pytorch again

MAli7319, 2025-03-12:
I started from exporting gcc and g++ paths again and creating the environment with pytorch. I mistakenly overlooked the part that installing the other cuda tools except from compiler. After this command:

`conda install cuda -c nvidia/label/cuda-12.4`

and installing the simpleknn:

`pip install git+https://github.com/VladimirYugay/simple-knn.git@c7e51a06a4cd84c25e769fee29ab391fe5d5ff8d`

finally it worked. Thanks a lot. I guess the next step is installing the rasterizer as the same way we did in the simpleknn.

VladimirYugay, 2025-03-12:
It's great that it worked! Now, installing the rasterizer should not be a problem. After that, continue installing the rest of the libraries manually.

I will close it for now, feel free to re-open it if you have any further questions.

MAli7319, 2025-03-12:
One think to remember that we will also need faiss-gpu to run the experiment. And faiss-gpu can only suitable with CUDA 11.4 and 12.1. That's why we need to install faiss-gpu-cuvs which is compatible with CUDA 11.8 and 12.4. Command:

`conda install -c pytorch -c nvidia -c rapidsai -c conda-forge libnvjitlink faiss-gpu-cuvs=1.10.0`

> Reference: https://github.com/facebookresearch/faiss/blob/main/INSTALL.md#user-content-fn-1-0a8abd26faa5cd0fb208aa9ed5812c6f

MAli7319, 2025-03-12:
I guess there is one missing point to run the project. Because I m receiving this error message:

> Traceback (most recent call last):                                              
>   File "/home/masenturk/Gaussian-SLAM/run_slam.py", line 9, in <module>         
>     from src.evaluation.evaluator import Evaluator                              
>   File "/home/masenturk/Gaussian-SLAM/src/evaluation/evaluator.py", line 25, in <module>
>     from src.evaluation.evaluate_reconstruction import evaluate_reconstruction
>   File "/home/masenturk/Gaussian-SLAM/src/evaluation/evaluate_reconstruction.py", line 9, in <module>
>     from evaluate_3d_reconstruction import run_evaluation
> ModuleNotFoundError: No module named 'evaluate_3d_reconstruction'

MAli7319, 2025-03-12:
I have literally overlooked the easiest part. Checked and uploaded, thanks again.
By the way, my reference has already pointed the same documentation :)

closed 2025-03-12 by MAli7319

MAli7319, 2025-03-12:
May I ask something about the runtime of the execution?
For example, I m running an experiment on the Replica/office0 data. It has 2000 frames recorded. How much time should it require to extract the map? I have RTX 3050 GPU and i5-12450H CPU.

VladimirYugay, 2025-03-12:
Sure, you can make a rough estimate from the paper. The system is approximately 1FPS, keep in mind that it is on powerful hardware.