VladimirYugay/Gaussian-SLAM #48: Anyway to avoid CUDA OOM when Merging segments?

lemonci, 2025-03-12:
Hello, thanks for your great work! (And I am looking forward to seeing your code for the multi-agent repo~)

I am now using a 4070 to run
`python run_slam.py configs/Replica/room0.yaml --input_path /home/datasets/Replica/room0 --output_path output/Replica/room0`
and encountered a CUDA OOM error during evaluation:

submap = torch.load(submap_path, map_location=self.device)
100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 39/39 [18:20<00:00, 28.22s/it]
TriangleMesh with 1778389 points and 3445621 triangles.
Mesh cleaning (before/after), vertices: 1778389/1734680, faces: 3445621/3412288
Align:  True

```
===========================
Evaluating room0
===========================
Could not run reconstruction eval
Traceback (most recent call last):
  File "/home/mistlab/Documents/ml/Gaussian-SLAM/src/evaluation/evaluator.py", line 267, in run
    self.run_reconstruction_eval()
  File "/home/mistlab/Documents/ml/Gaussian-SLAM/src/evaluation/evaluator.py", line 205, in run_reconstruction_eval
    evaluate_reconstruction(file_name,
  File "/home/mistlab/Documents/ml/Gaussian-SLAM/src/evaluation/evaluate_reconstruction.py", line 272, in evaluate_reconstruction
    result_3d = run_evaluation(
  File "/home/mistlab/anaconda3/envs/gslam/lib/python3.10/site-packages/evaluate_3d_reconstruction/evaluate_3d_reconstruction.py", line 131, in run_evaluation
    transformation = get_align_transformation(pred_ply_path, gt_ply_path)
  File "/home/mistlab/anaconda3/envs/gslam/lib/python3.10/site-packages/evaluate_3d_reconstruction/evaluate_3d_reconstruction.py", line 54, in get_align_transformation
    o3d_gt_mesh = trimesh.load(gt_meshfile)
  File "/home/mistlab/anaconda3/envs/gslam/lib/python3.10/site-packages/trimesh/exchange/load.py", line 111, in load
    loaded = load_scene(
  File "/home/mistlab/anaconda3/envs/gslam/lib/python3.10/site-packages/trimesh/exchange/load.py", line 193, in load_scene
    arg = _parse_file_args(
  File "/home/mistlab/anaconda3/envs/gslam/lib/python3.10/site-packages/trimesh/exchange/load.py", line 624, in _parse_file_args
    raise ValueError(f"string is not a file: `{file_obj}`")
ValueError: string is not a file: `data/Replica-SLAM/cull_replica/room0.ply`
Running global map evaluation...
Merging segments
  0%|                                                                                                                                                            | 0/39 [00:00<?, ?it/s]/home/mistlab/Documents/ml/Gaussian-SLAM/src/evaluation/evaluate_merged_map.py:72: FutureWarning: You are using `torch.load` with `weights_only=False` (the current default value), which uses the default pickle module implicitly. It is possible to construct malicious pickle data which will execute arbitrary code during unpickling (See https://github.com/pytorch/pytorch/blob/main/SECURITY.md#untrusted-models for more details). In a future release, the default value for `weights_only` will be flipped to `True`. This limits the functions that could be executed during unpickling. Arbitrary objects will no longer be allowed to be loaded via this mode unless they are explicitly allowlisted by the user via `torch.serialization.add_safe_globals`. We recommend you start setting `weights_only=True` for any use case where you don't have full control of the loaded file. Please open an issue on GitHub for any issues related to this experimental feature.
  gaussian_params = torch.load(submap_path)["gaussian_params"]
100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 39/39 [03:42<00:00,  5.70s/it]
Downsampled point cloud to 788641 points
Number of added points:  777467
Refinement:   8%|██████████▏                                                                                                                        | 781/10000 [00:35<06:57, 22.08it/s]
Could not run global map eval
Traceback (most recent call last):
  File "/home/mistlab/Documents/ml/Gaussian-SLAM/src/evaluation/evaluator.py", line 273, in run
    self.run_global_map_eval()
  File "/home/mistlab/Documents/ml/Gaussian-SLAM/src/evaluation/evaluator.py", line 218, in run_global_map_eval
    refined_merged_gaussian_model = refine_global_map(merged_cloud, training_frames, 10000)
  File "/home/mistlab/Documents/ml/Gaussian-SLAM/src/evaluation/evaluate_merged_map.py", line 129, in refine_global_map
    total_loss.backward()
  File "/home/mistlab/anaconda3/envs/gslam/lib/python3.10/site-packages/torch/_tensor.py", line 581, in backward
    torch.autograd.backward(
  File "/home/mistlab/anaconda3/envs/gslam/lib/python3.10/site-packages/torch/autograd/__init__.py", line 347, in backward
    _engine_run_backward(
  File "/home/mistlab/anaconda3/envs/gslam/lib/python3.10/site-packages/torch/autograd/graph.py", line 825, in _engine_run_backward
    return Variable._execution_engine.run_backward(  # Calls into the C++ engine to run the backward pass
  File "/home/mistlab/anaconda3/envs/gslam/lib/python3.10/site-packages/torch/autograd/function.py", line 307, in apply
    return user_fn(self, *args)
  File "/home/mistlab/anaconda3/envs/gslam/lib/python3.10/site-packages/gaussian_rasterizer/__init__.py", line 157, in backward
    grad_means2D, grad_colors_precomp, grad_opacities, grad_means3D, grad_cov3Ds_precomp, grad_sh, grad_scales, grad_rotations = _C.rasterize_gaussians_backward(
torch.OutOfMemoryError: CUDA out of memory. Tried to allocate 122.00 MiB. GPU 0 has a total capacity of 11.72 GiB of which 45.12 MiB is free. Including non-PyTorch memory, this process has 11.67 GiB memory in use. Of the allocated memory 10.44 GiB is allocated by PyTorch, and 918.44 MiB is reserved by PyTorch but unallocated. If reserved but unallocated memory is large try setting PYTORCH_CUDA_ALLOC_CONF=expandable_segments:True to avoid fragmentation.  See documentation for Memory Management  (https://pytorch.org/docs/stable/notes/cuda.html#environment-variables)
All done.✨
```

May I ask if there is some advice to manage the VRAM better (like clear cuda cache in some place) so that the evaluation can be finished? Thanks.

VladimirYugay, 2025-03-13:
Thanks for your interest in our work! 

This merging implementation is indeed VRAM-hungry. Here are a couple of things you can do to make it lighter:

1. Clean out the VRAM before running the merging and evaluation. Sometimes, the VRAM is not deallocated before the merging, which leads to huge consumption. By loading already computed sub-maps, you can also run merging and evaluation as a standalone process without the need to repeat the whole sequence

2. Tune these hyperparameters `new_submap_points_num: 600000`, `new_submap_gradient_points_num: 50000`. I believe you can reduce them 2 times without much of a negative effect on Replica. This will make the amount of Gaussians in the sub-map significantly smaller

3. Wait until the release of MAGiC-SLAM. I'm currently pushing the refactoring as much as possible. I hope to release it sometime next week. It consumes significantly less memory

lemonci, 2025-03-13:
> Thanks for your interest in our work!
>
> This merging implementation is indeed VRAM-hungry. Here are a couple of things you can do to make it lighter:
>
>     1. Clean out the VRAM before running the [...]

Hello, following your advice. I have added `torch.cuda.empty_cache()` in [Line 205 of evaluator.py](https://github.com/VladimirYugay/Gaussian-SLAM/blob/eaec10d73ce7511563882b8856896e06d1f804e3/src/evaluation/evaluator.py#L205) and changed the config to `new_submap_points_num: 150000`, `new_submap_gradient_points_num: 12500`.

However I still get the CUDA OOM error:

```
Running global map evaluation...
Merging segments
  0%|                                                                                                                                                                    | 0/39 [00:00<?, ?it/s]/home/mistlab/Documents/ml/Gaussian-SLAM/src/evaluation/evaluate_merged_map.py:72: FutureWarning: You are using `torch.load` with `weights_only=False` (the current default value), which uses the default pickle module implicitly. It is possible to construct malicious pickle data which will execute arbitrary code during unpickling (See https://github.com/pytorch/pytorch/blob/main/SECURITY.md#untrusted-models for more details). In a future release, the default value for `weights_only` will be flipped to `True`. This limits the functions that could be executed during unpickling. Arbitrary objects will no longer be allowed to be loaded via this mode unless they are explicitly allowlisted by the user via `torch.serialization.add_safe_globals`. We recommend you start setting `weights_only=True` for any use case where you don't have full control of the loaded file. Please open an issue on GitHub for any issues related to this experimental feature.
  gaussian_params = torch.load(submap_path)["gaussian_params"]
100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 39/39 [01:50<00:00,  2.82s/it]
Downsampled point cloud to 827347 points
Number of added points:  815319
Refinement:   8%|██████████▋                                                                                                                                | 771/10000 [00:36<07:22, 20.86it/s]
Could not run global map eval
Traceback (most recent call last):
  File "/home/mistlab/Documents/ml/Gaussian-SLAM/src/evaluation/evaluator.py", line 274, in run
    self.run_global_map_eval()
  File "/home/mistlab/Documents/ml/Gaussian-SLAM/src/evaluation/evaluator.py", line 219, in run_global_map_eval
    refined_merged_gaussian_model = refine_global_map(merged_cloud, training_frames, 10000)
  File "/home/mistlab/Documents/ml/Gaussian-SLAM/src/evaluation/evaluate_merged_map.py", line 129, in refine_global_map
    total_loss.backward()
  File "/home/mistlab/anaconda3/envs/gslam/lib/python3.10/site-packages/torch/_tensor.py", line 581, in backward
    torch.autograd.backward(
  File "/home/mistlab/anaconda3/envs/gslam/lib/python3.10/site-packages/torch/autograd/__init__.py", line 347, in backward
    _engine_run_backward(
  File "/home/mistlab/anaconda3/envs/gslam/lib/python3.10/site-packages/torch/autograd/graph.py", line 825, in _engine_run_backward
    return Variable._execution_engine.run_backward(  # Calls into the C++ engine to run the backward pass
  File "/home/mistlab/anaconda3/envs/gslam/lib/python3.10/site-packages/torch/autograd/function.py", line 307, in apply
    return user_fn(self, *args)
  File "/home/mistlab/anaconda3/envs/gslam/lib/python3.10/site-packages/gaussian_rasterizer/__init__.py", line 157, in backward
    grad_means2D, grad_colors_precomp, grad_opacities, grad_means3D, grad_cov3Ds_precomp, grad_sh, grad_scales, grad_rotations = _C.rasterize_gaussians_backward(
torch.OutOfMemoryError: CUDA out of memory. Tried to allocate 134.00 MiB. GPU 0 has a total capacity of 11.72 GiB of which 17.12 MiB is free. Including non-PyTorch memory, this process has 11.69 GiB memory in use. Of the allocated memory 10.36 GiB is allocated by PyTorch, and 1.01 GiB is reserved by PyTorch but unallocated. If reserved but unallocated memory is large try setting PYTORCH_CUDA_ALLOC_CONF=expandable_segments:True to avoid fragmentation.  See documentation for Memory Management  (https://pytorch.org/docs/stable/notes/cuda.html#environment-variables)
All done.✨
```

Any further suggestion?

lemonci, 2025-03-13:
Sidetalk, I am really curious about how you handled (1) the loop closure between submaps and (2)global map reconstruction in MAGiC-SLAM. Because:
In (1), the camera pose of each keyframe will be changed which means the point cloud in every submap needs to be changed, does it mean the 3D Gaussians in each submap needs to be trained again? If no, how do you handle the registration of each point cloud?
In (2), when there are some overlappings between the submaps of different agents, how do you decide the way to merge them?
Therefore I am looking forward to reading your code soon!

VladimirYugay, 2025-03-14:
1. How much VRAM do you have on your machine?
2. Can you leave all the settings the same and make this [value](https://github.com/VladimirYugay/Gaussian-SLAM/blob/eaec10d73ce7511563882b8856896e06d1f804e3/src/evaluation/evaluate_merged_map.py#L85) larger? For example, set it to 0.1.

Glad that you're curious about it!

lemonci, 2025-03-14:
> 1. How much VRAM do you have on your machine?
>
>     2. Can you leave all the settings the same and make this [...]

Hello I have a NVIDIA GeForce RTX 4070 according to `nvidia-smi` so it's 12 GB.
I am setting it to 0.1 to see if it works.

lemonci, 2025-03-14:
Hello, I think I still get CUDA OOM after adjusting the voxel size to 0.1:

```
Starting evaluation...🍺
Running trajectory evaluation...
ATE-RMSE: 0.62 cm, ATE-RMSE (aligned): 0.31 cm
Running rendering evaluation...
/home/mistlab/anaconda3/envs/gslam/lib/python3.10/site-packages/torchmetrics/functional/image/lpips.py:326: FutureWarning: You are using `torch.load` with `weights_only=False` (the current default value), which uses the default pickle module implicitly. It is possible to construct malicious pickle data which will execute arbitrary code during unpickling (See https://github.com/pytorch/pytorch/blob/main/SECURITY.md#untrusted-models for more details). In a future release, the default value for `weights_only` will be flipped to `True`. This limits the functions that could be executed during unpickling. Arbitrary objects will no longer be allowed to be loaded via this mode unless they are explicitly allowlisted by the user via `torch.serialization.add_safe_globals`. We recommend you start setting `weights_only=True` for any use case where you don't have full control of the loaded file. Please open an issue on GitHub for any issues related to this experimental feature.
  self.load_state_dict(torch.load(model_path, map_location="cpu"), strict=False)
  0%|                                                                                                                                                                    | 0/39 [00:00<?, ?it/s]/home/mistlab/Documents/ml/Gaussian-SLAM/src/evaluation/evaluator.py:86: FutureWarning: You are using `torch.load` with `weights_only=False` (the current default value), which uses the default pickle module implicitly. It is possible to construct malicious pickle data which will execute arbitrary code during unpickling (See https://github.com/pytorch/pytorch/blob/main/SECURITY.md#untrusted-models for more details). In a future release, the default value for `weights_only` will be flipped to `True`. This limits the functions that could be executed during unpickling. Arbitrary objects will no longer be allowed to be loaded via this mode unless they are explicitly allowlisted by the user via `torch.serialization.add_safe_globals`. We recommend you start setting `weights_only=True` for any use case where you don't have full control of the loaded file. Please open an issue on GitHub for any issues related to this experimental feature.
  submap = torch.load(submap_path, map_location=self.device)
100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 39/39 [00:10<00:00,  3.69it/s]
{'psnr': 38.10050503746335, 'lpips': 0.038791966296749796, 'ssim': 0.9911777250694506, 'depth_l1_train_view': 0.0028499253400838076, 'num_renders': 429}
Running reconstruction evaluation...
  0%|                                                                                                                                                                    | 0/39 [00:00<?, ?it/s]/home/mistlab/Documents/ml/Gaussian-SLAM/src/evaluation/evaluator.py:170: FutureWarning: You are using `torch.load` with `weights_only=False` (the current default value), which uses the default pickle module implicitly. It is possible to construct malicious pickle data which will execute arbitrary code during unpickling (See https://github.com/pytorch/pytorch/blob/main/SECURITY.md#untrusted-models for more details). In a future release, the default value for `weights_only` will be flipped to `True`. This limits the functions that could be executed during unpickling. Arbitrary objects will no longer be allowed to be loaded via this mode unless they are explicitly allowlisted by the user via `torch.serialization.add_safe_globals`. We recommend you start setting `weights_only=True` for any use case where you don't have full control of the loaded file. Please open an issue on GitHub for any issues related to this experimental feature.
  submap = torch.load(submap_path, map_location=self.device)
100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 39/39 [18:09<00:00, 27.95s/it]
TriangleMesh with 1802117 points and 3482985 triangles.
Mesh cleaning (before/after), vertices: 1802117/1750775, faces: 3482985/3440720
Align:  True

===========================
Evaluating room0
===========================
Rigid Transform Applied to Reconstructed Mesh:  [[ 1.    -0.001 -0.001 -0.003]
 [ 0.001  1.    -0.    -0.005]
 [ 0.001  0.     1.     0.004]
 [ 0.     0.     0.     1.   ]]
[Evaluate]
[compute distance from source to target]
[compute distance from target to source]
[Add color coding to predicted mesh to visualize error]
[Written to  /home/mistlab/Documents/ml/Gaussian-SLAM/output/Replica/room0/mesh/evaluation_results/room0.precision.ply ]
[Add color coding to target mesh to visualize error]
[Written to  /home/mistlab/Documents/ml/Gaussian-SLAM/output/Replica/room0/mesh/evaluation_results/room0.recall.ply ]
[get_f1_score_histo]
==============================
evaluation result : room0
==============================
distance tau : 0.010
precision : 0.9075
recall : 0.8362
f-score : 0.8704
==============================
precision statistics
min: 0.0001
max: 0.1130
mean: 0.0062
median: 0.0053
std: 0.0048
==============================
recall statistics
min: 0.0000
max: 0.6697
mean: 0.0276
median: 0.0047
std: 0.0789
[... cut ...]
torch.OutOfMemoryError: CUDA out of memory. Tried to allocate 20.00 MiB. GPU 0 has a total capacity of 11.72 GiB of which 19.12 MiB is free. Including non-PyTorch memory, this process has 11.69 GiB memory in use. Of the allocated memory 10.61 GiB is allocated by PyTorch, and 772.92 MiB is reserved by PyTorch but unallocated. If reserved but unallocated memory is large try setting PYTORCH_CUDA_ALLOC_CONF=expandable_segments:True to avoid fragmentation.  See documentation for Memory Management  (https://pytorch.org/docs/stable/notes/cuda.html#environment-variables)
```

VladimirYugay, 2025-03-14:
Interestingly, it crashes on 8%. With 0.1, this should not happen. Can you do Nvidia-smi right before the start of the refinement?

lemonci, 2025-03-14:
> Interestingly, it crashes on 8%. With 0.1, this should not happen. Can you do Nvidia-smi right before the start of the refinement?

I just added a breakpoint [here](https://github.com/VladimirYugay/Gaussian-SLAM/blob/eaec10d73ce7511563882b8856896e06d1f804e3/src/evaluation/evaluator.py#L218) and launched `run_evaluation.py` in debug mode.

Here we go at the breakpoint:

```
mistlab@tuf:~$ nvidia-smi
Fri Mar 14 19:44:55 2025       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 550.127.08             Driver Version: 550.127.08     CUDA Version: 12.4     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 4070        Off |   00000000:01:00.0 Off |                  N/A |
| 33%   46C    P8             19W /  200W |     677MiB /  12282MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
                                                                                         
+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A      2811      G   /usr/bin/gnome-shell                            2MiB |
|    0   N/A  N/A   1263083      C   ...lab/anaconda3/envs/gslam/bin/python        666MiB |
+-----------------------------------------------------------------------------------------+
```

VladimirYugay, 2025-03-15:
I see. I suspect the training frames' size is too large for the VRAM. 

Every refinement frame consists of several cuda tensors, as can be seen [here](https://github.com/VladimirYugay/Gaussian-SLAM/blob/eaec10d73ce7511563882b8856896e06d1f804e3/src/evaluation/evaluate_merged_map.py#L112). When refining, for every iteration, can you try loading a training frame to Cuda from scratch?

For example, during refinement, you iterate over the dataset object, and on every iteration, you load the data of the frame like this:
```
        color = (torch.from_numpy(
            self.dataset[idx][1]) / 255.0).float().to(self.device)
        depth = torch.from_numpy(self.dataset[idx][2]).float().to(self.device)
        estimate_c2w = self.render_poses[idx]
        estimate_w2c = np.linalg.inv(estimate_c2w)
        frame = {
            "frame_id": idx,
            "color": color,
            "depth": depth,
            "render_settings": get_render_settings(
                self.width, self.height, self.dataset.intrinsics, estimate_w2c)
        }
```

lemonci, 2025-03-15:
Hi, I am able to construct the frame under the function [run_global_map_eval](https://github.com/VladimirYugay/Gaussian-SLAM/blob/eaec10d73ce7511563882b8856896e06d1f804e3/src/evaluation/evaluator.py#L210) according to your hint. However, I would like make sure:

1. It should be `estimate_c2w = self.estimate_c2w[idx]` instead of `estimate_c2w = self.render_poses[idx]`?
2. Regarding the `idx` number itself:
(a) In the [comment](https://github.com/VladimirYugay/Gaussian-SLAM/blob/eaec10d73ce7511563882b8856896e06d1f804e3/src/evaluation/evaluate_merged_map.py#L94), you mentioned `"""Refines a global map based on the merged point cloud and training keyframes frames."""`. So I supposed you only used one frame per every five.
(b) However, it seems that the length of `training_frames` is 1000 in [run_global_map_eval](https://github.com/VladimirYugay/Gaussian-SLAM/blob/eaec10d73ce7511563882b8856896e06d1f804e3/src/evaluation/evaluator.py#L215)
(c) Also, when I put a break point just after `training_frame = next(training_frames) `in [refine_global_map](https://github.com/VladimirYugay/Gaussian-SLAM/blob/eaec10d73ce7511563882b8856896e06d1f804e3/src/evaluation/evaluate_merged_map.py#L111) and run 5 iterations. It seems that the frame ids are `1610, 542, 1818, 590, 366`. So they are not indexed every 5th frame.
So I would like to ask:
(1) How did you select the 1000 frames in the dataset, as they are not keyframes (there are only 800 keyframes if every fifth is selected)? 
(2) I suppose I need to iterate over the 1000 frames for 10 times as `max_iterations = 10000`?

VladimirYugay, 2025-03-15:
The logic is defined in the `RenderFrames` dataset. For every iteration, you can randomly pick a frame ID for refinement.

lemonci, 2025-03-15:
> The logic is defined in the `RenderFrames` dataset. For every iteration, you can randomly pick a frame ID for refinement and use the estimated pose.
Ok I am following this and it seems loading one frame at a time will makes `run_global_map_eval()` alone need about 86 hours to run...
I will see if I will still have OOM (probably not) and keep you updated....

lemonci, 2025-03-19:
Thanks now I do not have the out of memory problem. However, as it takes too much time. Eventually I am reproducing it on a server (so thanks for your slurm script).
You can kindly close this issue. Btw how is the code release plan for MAGiC? ;)